nebhead/PiTender #1: Once enabled, pumps do not turn off

rkabrick, 2020-06-06:
I'm having an issue where as soon as I trigger a drink to be poured, the right pumps become enabled however they never stop pouring. This continues even after I use the Cancel option in the GUI as well as when I stop the application from running on the pi.

nebhead, 2020-06-06:
Thanks for trying out my project!  Sorry things are not going so well.  I have to come clean and tell you that I haven't yet actually built the full project, but have tested with a Pi and some relays test most everything out.  A friend of mine was going to build out the hardware,  but hasn't had the time to complete it yet.  Regardless, I'm wondering if you can perhaps try to run the control.py manually and insert some debug print statements in the script to see if you ever get to the DeactivatePump() function.  Alternatively, you could add the following lines to control.py, in the DeactivatePump() function:

```
event = 'Deactivate Pump Function Successfully Called'
WriteLog(event)
```
Then, afterwards check your /log/events.log file.  

We can backtrack from there.

rkabrick, 2020-06-06:
`2020-06-06 19:35:00 Drink requested: coke
2020-06-06 19:40:56 Drink requested: tequila_sunrise
2020-06-06 19:40:56 Deactivate Pump Function Successfully Called
2020-06-06 19:40:56 Deactivate Pump Function Successfully Called`

At first I put the two lines in the beginning of the function and there was nothing in the logs. Then I added it to the body of the for loop and it printed to the log file. Hopefully this is what you were asking for

nebhead, 2020-06-07:
Ok, let's try this, where you see #DEBUGprint, let's change those to WriteLog instead.  For example:

```
#DEBUGprint('Pin number ' + str(pin_number) + ' initialized as output for ' + pump_number + '.  Set to 1. ')
```
to
```
WriteLog('Pin number ' + str(pin_number) + ' initialized as output for ' + pump_number + '.  Set to 1. ')
```
And for these lines the respective functions:

```
#DEBUGprint(pump_number + " Pump Activated. Dispensing " + ingredient)
#DEBUGprint(pump_number + " Pump De-Activated. Stopped dispensing " + ingredient)
```
to

```
WriteLog(pump_number + " Pump Activated. Dispensing " + ingredient)
WriteLog(pump_number + " Pump De-Activated. Stopped dispensing " + ingredient)
```

Then perhaps we can see if we are even getting to that point in the program.

nebhead, 2020-12-17:
I know this issue is quite old, but I think I discovered my logic bug that caused the pumps to be on.  I believe that I was not accounting for the relays to be active low (which they are) and was unintentionally activating all pumps at the same time.  

Anyway, I believe with the latest update, we should be all fixed up.  With that, I'm going to close this issue, but please do feel free to re-open if you ever see this occurring again.

onlinejannik, 2021-01-04:
Hey nebhead,

thanks a lot for this great project. At first I started to build the project as shown by HackerShack, but I didn't want the SPI Display and the Buttons did not work for me as I wanted (Because the GPIO inputs are switching from 0 to 1 as they want, without a pullup/pulldown resistor). So your project works great for me.

I use a pi zero with Raspbian Stretch. WebInterface works great and I'm able to change everything I need to change in the Web-UI. But when I start making a drink, all pumps are running and as mentioned before, they do not turn off. I tried reusing your debug mode you changed in your last commit and I can even see the loading bar on the screen - in normal mode I can't see it. 

What can I do to make my pumps work as I want? I use exactly the same Hardware like you/Hackershack.

Thanks!

onlinejannik, 2021-01-04:
Last but not least my logfile of control.py:

Traceback (most recent call last):
  File "/home/pi/pitender/control.py", line 192, in <module>
    main()
  File "/home/pi/pitender/control.py", line 174, in main
    PourDrink(status['control']['drink_name'])
  File "/home/pi/pitender/control.py", line 68, in PourDrink
    platform.ActivatePump(pump_number)
TypeError: ActivatePump() missing 1 required positional argument: 'settings'

nebhead, 2021-01-04:
That certainly looks like a bug to me. :)  Let me see what I messed up and try to upload a fix.

nebhead, 2021-01-04:
Wow, I'm really sorry, but somehow I completely borked the Raspberry Pi platform file... not sure what happened but I'm rewriting it now.  I'll try to test it later and re-post.  I must have been drinking too many cocktails when I pushed the last update. :P

nebhead, 2021-01-04:
Ok, I've updated the platform_raspi.py which _should_ work.  All you need to do is replace that file (or do a git pull) and try again.  Let me know how it goes on your end.  I actually don't have the full hardware to test with at the moment because it is with a friend - so it would be great if you could just drop me a note if it works (or doesn't).

onlinejannik, 2021-01-05:
Wow, thank you for your quick response. It works right now!
Only thing (just details) is the following log:

SyntaxError: invalid syntax
  File "/home/pi/pitender/control.py", line 67
    print(f"Pump number = {index}")
                                 ^
SyntaxError: invalid syntax
Traceback (most recent call last):
  File "/home/pi/pitender/control.py", line 33, in <module>
    from platform_raspi import PumpControl # Library for reading the ADC device
  File "/home/pi/pitender/platform_raspi.py", line 27
    print(f"Pin number {pin_number} initialized as output for {pump_number}.  Set to 1. ")

I removed the "f" and set the double quotation marks to single quotation marks (" to ') -> Now it works all fine. I will do a little test this evening (GMT+1 :D) and I will translate your code to german in the next couple of days.

nebhead, 2021-01-05:
Great!  Glad to hear that it is working for you now.  The Syntax Error you are getting appears to be a Python version mismatch?  I just test ran the control.py app with Python 2.x and got the same error, but with Python 3.x it's seemingly fine.   If you use the the install script to install the software, it _should_ run the scripts with Python 3.  

If you translate to German, I'd love to add a link to your fork or add it to a branch of this project!  That would be really cool!

onlinejannik, 2021-03-13:
Hey! It is working all well. I built the machine and would like to send you a few pictures and the german translated source code. Well, I am Not familiar with the github functions, how can I send you the pictures and the source Code?

nebhead, 2021-03-13:
Wonderful!  I'm excited to see it!  I Believe you can drop the images into the thread here.  And if you're not so comfortable pushing your changes to a repository on GitHub, you can zip up the files and paste the zip file here in this issue discussion.  It should be drag and drop into your browser when creating a response.  Otherwise, we can exchange e-mail and perhaps share a google drive link.

onlinejannik, 2021-03-13:
[image]

onlinejannik, 2021-03-13:
[image]

onlinejannik, 2021-03-14:
[image]
[image]
[image]
[image]
[image]
[image]

onlinejannik, 2021-03-14:
Yesterday we had the first test run and it worked all very well! The only thing I thought about is to get the pumps of one recipe running at the same time, not one after another. And I used an adjustable DC-DC step up converter, to give the pumps 24 Volts, because otherwise they are running really slow, I need to see . The actual rate is now 100 ml in 30s, not sufficient for use in a bar, but all enough for me and my friends. Great thing and really great code. Additionally I use now a Raspberry Pi zero, at first I used a Pi 4, but I think the 4 is a little bit too oversized for this project. Actually I use an iPhone Hotspot for the Pi and connect my iPad with the Hotspot to control everything. But currently I am thinking about a solution where the Pi zero has its own WiFi (I think this will work with an acces-point solution, but I'm not sure whether the zero supports this). I will try to copy the files during the next week from the SD Card. So Thank you a lot!!

nebhead, 2021-03-14:
> Yesterday we had the first test run and it worked all very well! The only thing I thought about is to get the pumps of one recipe running at the same time, not one after another. 

Yes!  Great suggestion!  I'm going to modify the code to run the pumps in parallel so drinks can be poured faster.  I just need to implement a little threading (just like the original code/project) and kick off multiple processes.  

> And I used an adjustable DC-DC step up converter, to give the pumps 24 Volts, because otherwise they are running really slow, I need to see whether the peristaltic pump heads get damaged with this high flow rate. The actual rate is now 100 ml in 30s, not sufficient for use in a bar, but all enough for me and my friends. 

Also, I've heard that even the same spec'd pumps may have different flow rates from each other - thus I am also looking at creating flow-rate settings for each individual pump.  Let me know how your testing goes with increasing the voltage. 

> Great thing and really great code. Additionally I use now a Raspberry Pi zero, at first I used a Pi 4, but I think the 4 is a little bit too oversized for this project. 

Thank you!  I just want to commend you on your build.  Your woodworking is gorgeous and the design is amazing!!!  Really nice job with that.  You are putting everyone else to shame. :)  

> But currently I am thinking about a solution where the Pi zero has its own WiFi (I think this will work with an acces-point solution, but I'm not sure whether the zero supports this). I will try to copy the files during the next week from the SD Card. So Thank you a lot!!

One thing you can do is to equip a HDMI screen on the PiZero and just launch a browser which goes to localhost (127.0.0.1).  The other option may be to use your iPhone/iPad/Android device to direct connect to the PiZero over wifi (https://raspberrypi.stackexchange.com/questions/117238/connect-android-smartphone-with-wi-fi-direct-to-a-raspberry-pi).  Then similarly, go to the pi's IP from that device.

onlinejannik, 2021-03-14:
Thank you for your reply! OK, sounds nice if you would like to modify the code to get the pumps running parallel.

> Also, I've heard that even the same spec'd pumps may have different flow rates from each other - thus I am also looking at creating flow-rate settings for each individual pump. Let me know how your testing goes with increasing the voltage.

Well it doesn't matter if the pumps have different flow rates. I will give you a feedback how the tests are running. I think you know that testing a cocktail machine isn't the badest thing :D Maybe during this test phase we can think about another method to insert the pumping time, because pumping time in seconds always needs to be calculated. Maybe the user is able to tell the Pi the flow rate of the pumps and can insert the pump flow rate in ml for every recipe, but I'm not sure how the intern calculating ml -> seconds can be done. Calculating (more or less) during drinking can be risky :D

> One thing you can do is to equip a HDMI screen on the PiZero and just launch a browser which goes to localhost (127.0.0.1). The other option may be to use your iPhone/iPad/Android device to direct [...]

Great idea! I will go through this and test it, sounds nice! I also thought about maybe insert a battery mode, maybe with batteries from cordless screwdrivers to keep the work for the project going :)

> Thank you! I just want to commend you on your build. Your woodworking is gorgeous and the design is amazing!!! Really nice job with that. You are putting everyone else to shame. :)

Thank you for the commend. I just thought that a machine like this needs a nice casing :-)
I would say let's stay in contact to discuss a few possible solutions and give other users the possiblity to build their own machine.

nebhead, 2021-03-14:
> Thank you for your reply! OK, sounds nice if you would like to modify the code to get the pumps running parallel.

OK, updated the code to adopt the original method of using threading to dispense all of the ingredients simultaneously.  See the latest commit.  This should speed up dispensing greatly especially for drinks with many ingredients.  Just pull the latest.  

> Well it doesn't matter if the pumps have different flow rates. I will give you a feedback how the tests are running. I think you know that testing a cocktail machine isn't the badest thing :D Maybe [...]

Yes, in the latest release, I've basically changed the pump time to instead volume (ml).  So that should help remove the user from calculating anything.  The only thing that needs to be calculated/calibrated is the flow rate of the pump initially, but once that is set, it's just ml.  

> Great idea! I will go through this and test it, sounds nice! I also thought about maybe insert a battery mode, maybe with batteries from cordless screwdrivers to keep the work for the project going :)

Sounds excellent!  

> Thank you for the commend. I just thought that a machine like this needs a nice casing :-)
> I would say let's stay in contact to discuss a few possible solutions and give other users the possiblity to build their own machine.

I'll setup a discussions tab on this repository so we can continue the discussions of new features, etc. there.  For now, I'll close out this issue thread.  Thanks again!

closed 2021-03-14 by nebhead