nebhead/PiTender #2: Configure Drinks on the Web Interface

Mavajax, 2020-11-30:
Is ist possible to implement a way to configure the Drinks in the Web Interface ?

nebhead, 2020-11-30:
Absolutely - This is definitely a feature I want to add in the future.  I'll work on this, but I have some other projects in the queue first so it might take a while.

nebhead, 2020-12-17:
I'm happy to report that I have updated this project and created a Recipe Builder page to create new drinks and add ingredients to the database!!  I've also squashed a few other bugs and made some other big improvements (like bringing the whole project up to Python 3.x).  
[image]

closed 2020-12-17 by nebhead

Mavajax, 2021-01-06:
Wow thank you for the Update !