nebhead/PiTender #3: I have built the original Project on Hackster now i am looking at your project!

Microgeek1963, 2021-02-16:
I love the front end on this!
I also liked the automated install very simple.

I created the project on Hackster and I ran into a few issues which led me to strip out the Lights and small screen they used I left the buttons.

I created a second setup for testing.....

I have an issue where after testing the pumps I determined none of them 9 so far 8 on the project and one in my testbed actually runs at 100ml per minute and each pump is slightly different. between about 69 to 73 ml per minute.

in the original project they used one variable "FLOW_RATE = 60.0/100.0" which I modified to "60.00/71.34"
an average of the actual flow rate for the eight pumps.

I am looking at your code and have not found any place where you have a variable for the flow rate?

Ideally after my testing I would love to see an additional value for this on each pump so when the drinks are poured. 
the pumps would be able to pump the correct amounts for each pump.

Microgeek1963

nebhead, 2021-02-16:
Hi @Microgeek1963, 

Glad you like the project - it was definitely fun to make and I learned a lot in the process.  So, I think we definitely can look at adding a flow-rate setting to this project to try and correct for varying pump flow rates.  Sounds like a very useful feature.  I'll take a look at this when I get a chance, though it may take me a little while to get to it.  Stay tuned.  

Thanks,
Ben.

nebhead, 2021-02-24:
OK, I've added a quick & dirty global flow rate value which can be accessed and edited in the settings/admin menu.  Seemed to work for me in my testing, but if you can pull the latest, give it a go, and let me know that would be wonderful.  

https://github.com/nebhead/PiTender/commit/298aa6d7b1149b62a6c6f094dbeb704de60ee4f7

If this seems to work OK for you, in the future I will create a per-pump flow control setting, so these can be tuned appropriately.  

Thanks!  
Ben.

Microgeek1963, 2021-02-26:
I am in the middle of updating the hardware on my bartender.I will test as soon as I can!

Microgeek1963, 2021-03-01:
Hello nebhead,

I have finished upgrading my hardware and run tests and found it works fine...

Thanks

Microgeek1963

Microgeek1963, 2021-03-01:
Hello Again,

I am a little surprised about the deviations between the pumps!

nebhead, 2021-03-01:
> I have finished upgrading my hardware and run tests and found it works fine...

Thank you!!!  I'll get around to making a per-pump flow control setting in the future so it's configurable per pump.  Thanks for testing!

nebhead, 2021-03-01:
> I am a little surprised about the deviations between the pumps!

Yeah, that's surprising!  Perhaps with more expensive pumps there is a smaller deviation.  Anyway, I'll close this one for now and will look at improving this feature in the future.

closed 2021-03-01 by nebhead