nebhead/PiTender #6: Pumps do not start

onlinejannik, 2021-03-15:
Hey!

I tried to update the PiTender with the new code including great ideas! But since I uploaded your files, the pumps are not running. Even the loading bar isn't displayed. At first I just uploaded the changed files, but even when I load the whole repository the pumps won't work. Maybe this is because of the updated flow rate section? 
Because I was about to translate the modified files :D

Do you know why this issue occurs?
Thanks!

nebhead, 2021-03-15:
I think I have a bug in the code.  I accidentally removed a line that sets prototype mode True or False at the beginning of the control.py file.  Let me fix and push the update.  Sorry for the inconvenience!

nebhead, 2021-03-15:
Let me take a look and see what's going on.   In the meantime, can you maybe try to backup and delete your settings.json file?  Since I added the flow control, it's possible that your settings file is out of date and is causing control.py to fail.  Also, maybe check your logs dir and let me know if there is an err log for control.py

onlinejannik, 2021-03-16:
Okay, I deleted the settings.json file and re-update it from github. Same failure as before. These are my logfiles:

control.err:

`SyntaxError: invalid syntax
  File "/home/pi/pitender/control.py", line 41
    print(f' * Thread: {pump_number} Started for {waitTime} seconds.')
                                                                    ^
SyntaxError: invalid syntax`

It prints this multiple times in the log file, after I deleted it and restarted the cleaning program for function test.

And the webapp.err, i don't know whether this is relevant for you:

`[2021-03-16 17:44:56 +0000] [610] [INFO] Starting gunicorn 19.6.0
[2021-03-16 17:44:56 +0000] [610] [INFO] Listening at: http://127.0.0.1:8000 (610)
[2021-03-16 17:44:56 +0000] [610] [INFO] Using worker: sync
[2021-03-16 17:44:56 +0000] [620] [INFO] Booting worker with pid: 620
`

Sadly I am not a programmer and my python knowledge is really limited, so I hope you can get along with these information.
Thank you!

nebhead, 2021-03-16:
> `SyntaxError: invalid syntax File "/home/pi/pitender/control.py", line 41 print(f' * Thread: {pump_number} Started for {waitTime} seconds.') ^ SyntaxError: invalid syntax`

Hmm... this is odd, it's almost as if your Python version doesn't support this style of print statements.  Anyway, you can simply comment those print statements with the format of `print(f'... {...}')` by adding a # in front of that line.  

Hopefully that helps us get past these types of errors.  Those print statements are for debug only anyway.  However, I would also be concerned that maybe your Python version is not 3.6+.  Perhaps at the command line if you can check your python version by launching Python3.  

> `[2021-03-16 17:44:56 +0000] [610] [INFO] Starting gunicorn 19.6.0 [2021-03-16 17:44:56 +0000] [610] [INFO] Listening at: http://127.0.0.1:8000 (610) [2021-03-16 17:44:56 +0000] [610] [INFO] Using worker: sync [2021-03-16 17:44:56 +0000] [620] [INFO] Booting worker with pid: 620 `

Yeah, this is all normal.  But thanks for providing as well.  
 
> Sadly I am not a programmer and my python knowledge is really limited, so I hope you can get along with these information.

No worries!  Happy to help.  This is how we all learn.

onlinejannik, 2021-03-18:
Okay, finally it works. I just put a brand new image on my Pi zero and reinstalled everything. So I think that a clean refresh of the Pi OS is most often the solution for things like that. Now I will start an extended testing period and will come back to you if any new ideas for modifications show up :-)
Thank you!

closed 2021-03-18 by onlinejannik

nebhead, 2021-03-18:
Thank you for following up on this and I am glad you have it all working.  Hopefully the improved dispensing gets those drinks into your friends hands faster. :)  Look forward to improving on this project in the future!